TomWhitwell/RadioMusic #128: Time CV reading astronomical numbers

jrkenny5, 2017-10-25:
Module test firmware seems to run fine, but the 4th led is full on. Upon running a serial monitor in arduino, I am receiving a ludicrous number for the time cv value. While the others (including potentiometers) seem to go anywhere from 1 to 1023, the time cv value is set at something like 21,480,000. Also, the sine wave output is very noisy and barely audible, hopefully these two issues are related, I figured it could be a bridged solder connection but I can't seem to find one anywhere on my boards.

I also have installed the module into my rack, so it is well grounded. There are several other modules running fine on the same power supply. Additionally, i have installed the actual radiomusic hex and that also does not work properly, it again seems to be very low in the background (and i believe that the audio IS playing, albeit from the furthest start point and looping from there.) but it's so noisy i can't be certain.

jrkenny5, 2017-10-25:
Additionally, when turning the pots while running the module test, anything other than the extreme values (either 1 or 1023) seems to add a high pitched saw wave??? I'm extremely confused now.

jstamant, 2018-03-22:
@jrkenny5 I'm having the same issue. Running the module test sketch shows an exorbitantly high value for the time CV (>21,000,000). I'll be performing more troubleshooting on my build before opening an issue. There's still plenty of troubleshooting I can do before asking for help.

At a glance, I would think this out-of-range time CV reading would indicate something wrong with the Teensy, the module-test sketch, or the Teensy's version. But again, I have more troubleshooting to do.

I saw that issue #141 might be related. I can't comment any more on that, other than I might try the chord organ firmware to see how it runs.