TomWhitwell/RadioMusic #129: Ground noise

Zek75, 2017-10-29:
Hi, I built a radio music. Everything works fine except I have a buzz or ground noise as soon as the module plays or has something plugged in, like an lfo or whatever. Even is the out is not connected, I hear the noise. I tried to decrease the volume, but it didn't change anything. I also resoldered every joint, tried not to heat the diodes etc... When I touch the pins of the TL074 socket when it's playing, it makes a lot of noise. Do you think the problem can come from this IC? Do I need to change it? Great module by the way!

BobJack, 2018-02-22:
same problem... but at the same time the sound of the sample slowly fades away and little distort... i try to put 100k resostor, as in the issue #70  and write here, if something change

BobJack, 2018-02-24:
yes, i confirm that after adding 100k resistor - no problems in module (5 hours - ok)

Zek75, 2018-02-26:
Ok I will try this thanks!

tomterrien, 2019-02-11:
Hi ! having the same issue here , did it work for you Zek75 ?