TomWhitwell/RadioMusic #132: Problems compiling and uploading new firmware

MegaGlabrous, 2017-12-03:
I am attempting to update my radiomusic (which has been working perfectly for 18 months or so with the original firmware) to the latest. I downloaded Arduino 1.85 and Teensyduino with no apparent problems. But I am failing to Verify or Upload the new code.

Can anyone help?
Verify fails with:
[Verify.txt](https://github.com/TomWhitwell/RadioMusic/files/1524781/Verify.txt)

Upload fails with:
[Upload.txt](https://github.com/TomWhitwell/RadioMusic/files/1524782/Upload.txt)

TomWhitwell, 2017-12-03:
Hi, you don't have to compile anything to use the new firmware, just upload the .hex file with teensy loader. Tom

tomterrien, 2019-01-31:
Hi Tom , i'm Tom :) 
I'm trying to update radio Music's Firmware with the sampler one but teensy app says "unreadable" when i open the .hex file :/

Thonk-Steve, 2019-01-31:
Are you grabbing the firmware from here?
http://www.polyfather.com/radio_music/

'DOWNLOAD FIRMWARE' at the top of the page

Cheers

Steve
Thonk

tomterrien, 2019-01-31:
Maybe i'm not in the right topic ? :)

TomWhitwell, 2019-01-31:
You might be having trouble downloading the .hex file from github 

Try this link: https://raw.githubusercontent.com/jamescarruthers/RadioMusicAltFirmware/master/Sampler_1_0.hex
- then 'save as' and save as text
- then edit the file so it's called sampler_1_0.hex (remove any .txt extension) 
- Open the .hex file in a text editor - if you see a series of numbers after colons, that's correct.  If you see HTML code, that's not correct 

Upload that using the Teensy Loader app, NOT the Arduino IDE

tomterrien, 2019-02-02:
Found solution ! sorry for the late reply, i downloaded it zipped and it's fine , update done. Now i have to figure out ouut to be creative with this ^^ 
Thanks !

Tigersi76, 2020-05-02:
I'm late to the party and very much a newbie, can anyone help.

I'm looking to update my Radio Music to the 2017 firmware.
Do i have to have the module connected to my system whilst uploading or can i just take it out of the rack and upload it with my mac via usb?

Thank you so much for reading this and sorry if this seems like a dumb question to most of you.

Si

eeweegh, 2020-05-02:
it depends on whether the USB power trace on the teensy has been cut or not. You can find all of the info on https://github.com/TomWhitwell/RadioMusic/wiki/Preparing-the-Teensy

Tigersi76, 2020-05-03:
@eeweegh 
Thank you so much for your reply. I have read through that and already had but am still unsure. 
I feel daft for asking these questions as it’s probably simple stuff to you guys. 
Here are the pics of mine. I think there is no connection but would appreciate for someone who knows to confirm. 
Once again thank you for your time. 

[image]
[image]

eeweegh, 2020-05-03:
I think there *is* a connection, and you can just power it up and upgrade it, connecting it solely to USB. Look at your second picture, bottom right of the Teensy: there are three pads (a big one (VUSB) and two smaller ones). As far as I can judge, there is still a trace (thin copper line) between the two small pads.
[image]

Tigersi76, 2020-05-03:
Once again thank you so much. @MegaGlabrous this has sorted it. 
I now have .wav on my module. 
Very very much appreciated 

Si. :)

scheersvincent, 2021-02-06:
Hi!
im also having issues switching between chord organ firmware and radio music, i want to switch back to radio music firmware. 

when i attach the module with a usb cable to my macbook, the red light above the "station" know lights up, so i dont think the powerstrip was cut. 
However the LED on the teensy is never lighting up 

When i try to upload the firmware via the teensy loader, the program doesnt seem to respond, 

[image]

scheersvincent, 2021-02-06:
Also pressing the button on the teensy doesnt change anything

scheersvincent, 2021-02-06:
Pressing the physical button on the teensy only makes the red light above the "station" know stop lighting up

TomWhitwell, 2021-02-06:
Hi @scheersvincent there are a lot of tips here: 
https://www.pjrc.com/teensy/troubleshoot.html

I don't know if it's just the image you used, but that screenshot shows "blink_slow_TeensyLC.hex" - the file you need is "blink_slow_Teensy32.hex" because you're using a Teensy 3.1 or 3.2 (the same brain) 

Tom

scheersvincent, 2021-02-07:
Hi Tom,

Thanks for taking an interest, i appreciate it!

After everything i tried the chord organ mode still works fine in my case, which i think indicates that there is nothing wrong with the teensy itself. its just uploading new firmware where im stuck at.

Im hoping the issue is just that my usb-cable isnt made for data transfer but just for charging, i ordered a proper one and will keep you posted. 

Cheers,
Vincent

scheersvincent, 2021-02-09:
Fixed! 

Just had to use a proper usb cable, now installing new firmware works like a charm! 

thanks @TomWhitwell !