TomWhitwell/RadioMusic #133: Standard/musical tuning playback

benrossm, 2017-12-13:
Hey there, im considering picking up a Radio Music because of the new firmware, but I have some basic questions i was hoping someone could answer.. With 1v/oct pitch tracking, is it possible to play back something sampled in C musically (that is true to the pitch of the voltage coming in, not relative to the start knob position)? In the videos ive seen it was noted that the START knob has to be in the center for the sample to play back at the original pitch, but i dont think the start knob has a locking pot for dead center am i wrong? Because that would be hard to tune. How does this work? Is it possible?

Normalised, 2017-12-13:
Both CV and the Knob pitch controls can be quantised internally. The config file has options for both. That way you'll have a reasonable amount of travel between semitones on the knob control.

http://polyfather.com/radio_music/

benrossm, 2017-12-14:
thanks alot, ill give it a shot.