TomWhitwell/RadioMusic #135: Reseting each time the station is changed

aleszylder, 2018-02-06:
Hi! Well, the title is quite self explanatory, each time I change the station, using the knob or CV input, it reset... It even light the reset LED... 

Any clues?

hallundrauch, 2018-07-11:
hey aleszylder,
have you figured it out? got the same issue 
best, benjamin

Thonk-Steve, 2018-07-12:
> ..and my samples are not looping (looping is false (=0) on settings..) but
> still resetting each time I change the station by knob or cv

does turning the loop flag on change the behaviour?

hallundrauch, 2018-07-13:
no, unfortunately not. when I change looping to 1 the files are looping (so that is working) but either way samples are always reseting while changing the station..
I also tried to reset teensy with a new generated hex file from the editor, but still the same issue

Thonk-Steve, 2018-07-13:
Do you have a digital multimeter?

hallundrauch, 2018-07-14:
okey sorry, dont know why, but now its working that way. started with a new sd card from scratch. so samples are looping (1, true) and on "radio mode", so its not reseting anymore when I change the station. only when looping is false. is it supposed to work that way? but anyway its fine now :)
thanks for your time.

BBGuy, 2018-07-16:
My looping is on and it resets wen I change a channel. I assumed it is just the way it was. haven't got it next to me so will check what other settings I got. be great if it didn't

BBGuy, 2018-07-26:
I think I figured out what my issue was: 
In Pitch Mode it restarts each time you change the channel otherwise it plays in the background.  
Is this the corect behivier? Would love pitch mode + background playing.

alienlebarge, 2019-06-14:
Same here.
When `Looping=0`, turning the radio knob start the sample from the beginning.
When `Looping=1`, it behave like a radio. The sample keep playing in the background.

YanHart, 2019-09-29:
Hi! Same here.
As suggested by previous user, i would love to be able to have : looping on + pitch mode without resetting each time station is changed. It could maybe be another setting option?
Thanks if it can be implemented some days!