TomWhitwell/RadioMusic #137: Teensy not connecting

dedwen13, 2018-02-14:
Hey!
I recently got my hands on a second hand Music Thing Radio Music and for some strange reason teensy app doesn't find the teensy chip after pressing the program mode button. 

I had it set up in the rack before trying to program it and it seemed to work. I downloaded a firmware (Radio_Music_2017.2.hex) which seems to be too large for the chip as well. program and reboot mode are inactive.
I'm using a mac with OS 10.9.5. 
After calling the seller who was all nervous on the phone he told me that there needs to be some sort of bridge between some solder points which I believe is absolute bullshit.
any chance to reset the teensy chip in another way to get the latest firmware and the module working again?
best regards,

dedwen

BBGuy, 2018-02-14:
Hi  dedwen13 
The seller is corect. Checkout the section "Cut the USB power trace" on https://github.com/TomWhitwell/RadioMusic/wiki/Preparing-the-Teensy

Thonk-Steve, 2018-02-14:
> indeed the guy cut the usb power trace. thanx BBGuy!
> I appreciate your help! usb found and firmware updated.

have more faith ;)

dedwen13, 2018-02-14:
@Thonk-Steve hard if you got ripped couple of times but as you see I still try my best :)

TomWhitwell, 2018-02-14:
Always nice to have a happy ending

closed 2018-02-14 by TomWhitwell