TomWhitwell/RadioMusic #138: no sound

dedwen13, 2018-02-14:
after updating to the latest firmware via polyfather.com and putting wav files on the card there is no sound coming out of the module. module seems to respond tho.
any clues?

TomWhitwell, 2018-02-14:
Have you placed the Wav files in the right folders? https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-and-File-Structure

dedwen13, 2018-02-14:
since my first issue was a cut usb power by the previous owner I didn't recognise, I updated the module while powered up by a uZeus power supply so the teensy app would find the module.
firmware update went smooth. maybe I did a mistake there?
like I said the module seems to respond to reset triggers but doesn't output sound unfortunately.

dedwen13, 2018-02-17:
oh and the leds behave quite weird and random now when I turn the station knob. not like I read how they are supposed to behave.

oblivionratula, 2018-02-17:
Firmware: https://github.com/TomWhitwell/RadioMusic/tree/master/Collateral/hex%20files
"Latest" is what you're looking for, I think. Don't know if it'll help, but good luck.

jackpf, 2018-04-29:
I seem to be having the same issue - did you resolve it?