TomWhitwell/RadioMusic #140: Lights Blinking in Fast Succession when inserting SD card

electrojoeuk, 2018-02-25:
Module powers up fine (I think...). First LED comes on, then after 3 seconds it moves through each LED (i.e. 1, 2, 3, 4). However, as soon as I put an SD card in (I've used the Blank template file 0-15), the LEDs blink in fast succession. I'm thinking this might have something to do with the SD card holder? Hex file seems to have loaded fine... Anyone had and SOLVED a similar problem?

Going mad....

electrojoeuk, 2018-03-02:
So I tried re-soldering the card slot and touching up the solder in general, tried multiple brands of SD cards,  still same issue. However, I decided to load the Organ Hex file and its works great, although I didn't test the SD card feature. So as far as I can tell it must be something to do with the circuit path when reading the SD card...

Thonk-Steve, 2018-03-02:
Hey electrojoeuk, could you post good photos of both sides of both PCBs please?

electrojoeuk, 2018-03-03:
Hello Steve, as attached. Many thanks.
[image]
[image]
[image]
[image]

eeweegh, 2018-03-04:
just my 2c, I'd reflow a few of the components, so you get a nice solder bump at the component side. Not sure it's going to cure your problem, but at least you wouldn't have to worry about unreliable connections.

Also (and now in pedantic mode, so bear with me), give the board a good scrub with IPA to get the flux residue off: it looks nicer :-), but it helps you to better see a bad solder connection or a bridge.

electrojoeuk, 2018-07-27:
Was so fed up with debugging, decided to build a new one from scratch... Same problem. All the microSD cards I had been using were Class 10, decided to try a Class 4 microSD, worked like a charm!

closed 2018-07-27 by electrojoeuk

Thonk-Steve, 2018-07-27:
What brand were the class 10 cards?

electrojoeuk, 2018-07-28:
The ones I couldn't get to work were the following:
SanDisk Extreme - 32GB - Class 10
Integral - 16GB - Class 10
Team - 8GB - Class 10

Thonk-Steve, 2018-07-28:
Yeah I'm stumped, I've only ever used SanDisk class 10 cards since radio
music was released. Let us know if you run into any more problems.

francescoco, 2019-03-18:
@electrojoeuk
what's the green resistor on the back of the pcb for ?

simonjuha, 2019-03-18:
> @electrojoeuk
> what's the green resistor on the back of the pcb for ?
https://github.com/TomWhitwell/RadioMusic/issues/70