TomWhitwell/RadioMusic #142: Files not playing from SD card

joeypapz, 2018-03-21:
I just completed my build and was so stoked to test out the module, but the files are not playing.

At first, I thought it was an issue with the audio chain, but I loaded the Module Test (https://github.com/TomWhitwell/RadioMusic/tree/master/Collateral/ModuleTest) which proved that the module is working perfectly. Noise was coming out and the corresponding LEDs were changing with the inputs and pots. Then I uploaded the Chord Organ to give it a test run, which also worked perfectly (except I didn't have an SD card in).

Now I'm back to the Radio Music and I put my SD card in and I'm able to scroll through the banks on the SD card, and when rolling through the stations, the LED blinks to show that the stations are changing too, but the top 4 LEDs remain unlit and no sound is coming out except for some pops when I change banks or hit different stations when scrolling through.

Does anyone have any advice or suggestions? I've looked over my soldering and have re-soldered some stuff, but it all seems to not be helping.

Thank you!

Normalised, 2018-03-21:
Which radio music firmware and whats on the SD card?

joeypapz, 2018-03-21:
I have a bunch of RAW files I made using audacity to the proper specifications, then I also made a folder using the provided RAW files (vinyl/birds) and an additional folder with both RAW and WAV, just to add more variables. Nothing is working. Also, I downloaded the most recent firmware from http://polyfather.com/radio_music/  (Radio_Music_2017.2.hex)

[image]

Normalised, 2018-03-21:
And the 0,1 etc folders are in the root of the SD card?
Can you paste the config file in please?

joeypapz, 2018-03-22:
Thank you Normalised! The 0,1 etc folders were in a folder in the SD card. I just threw them all into the root of the SD card and now it works perfectly. It was making me so frustrated.

Thanks!

Normalised, 2018-03-22:
Great. Glad it works.

closed 2018-03-22 by Normalised