TomWhitwell/RadioMusic #145: Digital noise drowns the audio

simonjuha, 2018-06-03:
When i power up my RM i get some werid digital noise which drown the original audio, as soon as anything gets plugged into it. The noise stays even when unplugging the inputs.
The noise seems to disapear for at short time when i touch pins on the teensy with my finger.

I have checked all the solder joints and even resoldered som that seems worthy.

Thonk-Steve, 2018-06-03:
Hey, some questions:

Is it operating normally other than the noise?

What other modules are in the case? Anything with a lot of LEDS or a screen?

What power supply are you using?

simonjuha, 2018-06-03:
I have build two RM. The one of them works perfectly, the other one is the one with the problem. So i doubt it is something with the case or other modules.

I am using the doepfer A-100 diy kit.
No modules with screen or many leds.

also i tried switching the Teensy between the two modules.

Thonk-Steve, 2018-06-03:
Hi Simon, you've identified that  problem is solely down to the teensy then
when you swap then? The main modules have identical performance?

If not I'd suggest also trying combinations of the main two PCBs too.

simonjuha, 2018-06-03:
both modules are having the digital noise problem.
The noise comes when somthing in plugged into a input.
The noise remains even after turning on and off the whole rack.
The noise disapear when rack has been off for some time.

I soon have a new case finished. Will try to use the RM-modules in the new one, and see if the noise remain. Thinking it actually might be the case somehow.

simonjuha, 2018-06-03:
UPDATE:
I tried putting the RM-modules into another case; digital noise still acts the same way as before.
Tried putting the ouput from the modules into another mixer; same.

Thonk-Steve, 2018-06-03:
Yeah thats interesting.. it's a Doepfer PSU2 with the 'donut' toroidal
transformer?

simonjuha, 2018-06-03:
No. it's a AC to DC and a external 15v ac adatper.
[image]

I think i said that the noise disapear when i touch Teensy. It's actually seems to be pin 5 or something on the Tl074.

simonjuha, 2018-06-20:
Tried with SD-card rated class 10 instead of class 4. Noise are the same.

simonjuha, 2018-06-20:
Okay. i might have fixed the whole thing. having a 330k from pin 12 on the op amp to ground seems to keep the sneeky noise away :) would there be some other problems with doing this?

closed 2018-06-21 by simonjuha