TomWhitwell/RadioMusic #146: RM never worked: LED's are looping, then clicking and beeping sound (with or without SD card)

hallundrauch, 2018-07-11:
Hey all,

I finished Radio Music today and have a similar Issue as electrojoeuk in Issue#140, but a bit different:

Turning RM on, all Leds are blinking one after another (ca 3 sec / led), then theres a clicking&hi-freq beeping sound sound and it starts from the beginning. When I put the SD card in it is the same just much more faster (one "loop/beep" in a second). 

Checked all parts, the voltages (+12,-12, +5), teensy is working fine standalone with blinking test hex files. Installed the Radio_Music_2017.2.hex firmware.

So unfortunately this module never worked so far and I have no clue.
Is the IC damaged maybe? 

Best, Benjamin

hallundrauch, 2018-07-11:
Thanks so much for the quick response.
Was sure I formatted the right way, but seems I was wrong, re-formatted and the first sample is playing!

Thanks!!!

Hi Benjamin, this sounds like what happens when the card isn't formatted
correctly, did you follow these instructions for the folder structure etc?
https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-%26-File-Structure

Thonk-Steve, 2018-07-11:
perfect!

bassinfected, 2018-08-19:
Hey,
i have the same issue now.
The module was running for half an year at least and now i wanted to change some samples.
Since i replaced the samples it doesn´t work anymore and also keeps flashing etc.
I tried to format the SD card by SanDisc 5x now but still the same result.
It´s Mac OSX i am using but i set it to FAT every time.
Any idea what to do?

TomWhitwell, 2018-08-19:
Hi, sorry to hear that. 
Perhaps check the folder format on the card - it has to be arranged exactly as described here: https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-and-File-Structure

bassinfected, 2018-08-20:
I also tried the prepared SD folders you had but they also don´t work.
Updated the software too but it´s still the same issue.

bassinfected, 2018-08-25:
> Can you post a screenshot of the root of your file/folder structure with
> hidden files visible?

I tried the prepared one by you and a guy who postet a prepared card in a forum but.
 

Gesendet: Montag, 20. August 2018 um 17:20 Uhr