TomWhitwell/RadioMusic #148: Intermittent Audio Out

RTripp71, 2018-08-18:
I had this issue a few months ago and it seemed to resolve itself. Now it's back again: 

- One day the module works fine, and the next it appears fine, but no audio out. 
- Restart, but no change. Try that a few more times, still no audio.
- Loosen the module and check connections. The boards can wiggle a bit over time, so I make sure they are well connected. Still no audio.
- Check and resolder numerous points. Still no audio.
- The next day, I turn it on and audio is on. Working as it should, but I plug in a cable and audio is gone again. 
- Remove from case and reflow all solder. Still no audio.
- Re-install everything. No change.

I am coming to believe that there is a gremlin in here somewhere. I use and need this module for live performances. I like it, but am losing faith in it's reliability. It's been great for months a several live shows. Any ideas what might be going on?

Thanks in advance

Thonk-Steve, 2018-08-25:
> Well, after a week I put it back in the case and of course it works as
> normal. I guess I'll try the resistor trick as noted in the other thread
> and hope that prevents the issue. This just makes no sense to me at all. I
> will have to have a backup sample player (my disting) or replace this with
> something else. I had planned to put another one in my case too.

Maybe you maybe have some cold solder joints as it certainly sounds like a
hardware issue not software. They can be susceptible to changes in
temperature, and not present themselves at all at other times. That is
something it's possible to test at least.

RTripp71, 2018-08-26:
Thank you for the advice. I will go over the soldering again, but it's working now. :)