TomWhitwell/RadioMusic #149: Trying to put Radio Music onto my Chord Organ... please help

dimleaffall, 2018-09-30:
Arduino says #include <Audio.h> 

No such File or directory????

No clue man.

dimleaffall, 2018-09-30:
Arduino: 1.8.7 (Windows 10), TD: 1.44, Board: "Arduino/Genuino Uno"

In file included from C:\Users\Scott R\Desktop\RadioMusic-master\RadioMusic-master\RadioMusic\RadioMusic.ino:44:0:

AudioSystemHelpers.h:4:19: error: Audio.h: No such file or directory

compilation terminated.

exit status 1
Audio.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dimleaffall, 2018-09-30:
Arduino: 1.8.7 (Windows 10), TD: 1.44, Board: "Arduino/Genuino Uno"

In file included from sketch\AudioSystemHelpers.h:5:0,

                 from C:\Users\Scott R\Desktop\RadioMusic-master\RadioMusic-master\RadioMusic\RadioMusic.ino:44:

RamMonitor.h:85:20: error: malloc.h: No such file or directory

compilation terminated.

exit status 1
malloc.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

What does this all mean?  I know NOTHING.

dimleaffall, 2018-10-01:
Arduino: 1.8.7 (Windows 10), TD: 1.44, Board: "Arduino/Genuino Uno"

In file included from sketch\AnalogInput.cpp:14:0:

AnalogInput.h:13:20: error: expected ')' before 'pinIndex'

   AnalogInput(uint pinIndex);

                    ^

AnalogInput.cpp:23:25: error: expected constructor, destructor, or type conversion before '(' token

 AnalogInput::AnalogInput(uint pinIndex) {

                         ^

In file included from sketch\LedControl.cpp:2:0:

LedControl.h:28:3: error: 'elapsedMillis' does not name a type

   elapsedMillis bankFlashTimer = 0;

   ^

sketch\LedControl.cpp: In member function 'void LedControl::flash()':

LedControl.cpp:40:5: error: 'bankFlashTimer' was not declared in this scope

     bankFlashTimer = 0;

     ^

sketch\LedControl.cpp: In member function 'void LedControl::bankAndSingle(int, int)':

LedControl.cpp:49:12: error: 'bankFlashTimer' was not declared in this scope

         if(bankFlashTimer >= bankLedFlashTimes[bank]) {

            ^

In file included from sketch\SDPlayPCM.cpp:27:0:

SDPlayPCM.h:44:37: error: expected class-name before '{' token

 class SDPlayPCM: public AudioStream {

                                     ^

SDPlayPCM.h:42:23: error: 'AUDIO_BLOCK_SAMPLES' was not declared in this scope

 #define AUDIOBUFSIZE (AUDIO_BLOCK_SAMPLES * 27)

                       ^

sketch\SDPlayPCM.h:82:28: note: in expansion of macro 'AUDIOBUFSIZE'

  unsigned char audioBuffer[AUDIOBUFSIZE];

                            ^

sketch\SDPlayPCM.h: In constructor 'SDPlayPCM::SDPlayPCM()':

SDPlayPCM.h:46:20: error: class 'SDPlayPCM' does not have any field named 'AudioStream'

  SDPlayPCM(void) : AudioStream(0, NULL) { begin(); } 

                    ^

sketch\SDPlayPCM.cpp: In member function 'void SDPlayPCM::loopPlayback(bool)':

[... 699 lines cut ...]
C:\Users\Scott R\Desktop\RadioMusic-master\RadioMusic-master\RadioMusic\RadioMusic.ino: In function 'void peakMeter()':

RadioMusic:424:7: error: 'peakDisplayTimer' was not declared in this scope

  if( (peakDisplayTimer < 50) || (meterDisplayDelayTimer < settings.meterHide) ) return;

       ^

RadioMusic:424:34: error: 'meterDisplayDelayTimer' was not declared in this scope

  if( (peakDisplayTimer < 50) || (meterDisplayDelayTimer < settings.meterHide) ) return;

                                  ^

RadioMusic:431:2: error: 'peakDisplayTimer' was not declared in this scope

  peakDisplayTimer = 0;

  ^

expected ')' before 'pinIndex'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Normalised, 2018-10-01:
Do you just want your Chord Organ to be a Radio Music or do you want to customise the Radio Music code?

You can download the Radio Music firmware from here :

http://polyfather.com/radio_music/

And load that onto the module without compiling or editing code.

TomWhitwell, 2018-10-01:
You should NOT use the Arduino IDE to change the firmware. 
Download the firmware .hex files from: 

http://polyfather.com/radio_music/
http://www.polyfather.com/radio_music/

Then install with Teensy Loader, as described in the the instructions here: 
https://github.com/TomWhitwell/RadioMusic/wiki/Preparing-the-Teensy

closed 2018-10-01 by TomWhitwell

dimleaffall, 2018-10-01:
I feel like I'm SO close.  In the Teensy loader it says this: can't open
file 'C:\Users\Scott R\Desktop\Radio_Music_2017.2\Radio_Music_2017.2.elf'
(error 2: the system cannot find the file specified.)

TomWhitwell, 2018-10-02:
Sounds like you are close! I think you need a hex not an .elf - follow the instructions for teensy loader