TomWhitwell/RadioMusic #152: Reset trigger "pops" Audio Output

hankyates, 2018-12-03:
Hi there,

For whatever reason it sounds like DC is leaking into the audio output from the reset. Any advice for trouble shooting this? 

I suspect that maybe there is a capacitor or diode somewhere that isn't doing it's job, but I am unsure of which component.

hankyates, 2018-12-31:
Did some more investigation last night. I resoldered all capacitors which did not fix the popping. I noticed that the popping seems to be get louder as the audio signal from the DAC gets louder. Leading me to think this might be a software issue. I tried setting the "declick" setting but this didn't resolve the issue either.

Thonk-Steve, 2018-12-31:
Hi Hank,

Can you upload a direct recording of how this click sounds?

Thanks

Steve
Thonk

christophebruynseels, 2019-04-17:
Hi,
I am having the same problem. Whenever i reset the sample it clicks in and out. I tried to attenuate the cv but that didn t help, also tried different settings in the software but nothing seems to help.

chris

Thonk-Steve, 2019-04-17:
Hi Chris, same request as for Hank then, it's best if you can upload some
recordings. Also do you have an oscilloscope? It would be useful to check
you don't have a DC offset in your system which is a common cause of clicks.

hankyates, 2019-04-23:
Hey Steve I'll get you some audio but I did check with an oscilloscope
couldn't find any DC leaking. It's pretty strange because it doesn't appear
in the audio peak waveforms either.

mercurialization, 2020-06-09:
Has this issue ever been acknowledged/corrected? I'm having the same issue as well on a brand new build. I've set the crossfade as high as 1000ms in hopes of stopping the click/pop, but it doesn't affect the sound when 'Reset' is triggered. It makes the module a bit unusable for my case which is ambience.

hankyates, 2020-06-10:
@mercurialization could you post an audio sample?

jacoblysgaard, 2020-10-27:
I have the same problem. Here is an audio recording where it should be quite pronounced. 

https://soundcloud.com/lysgaardsounds/radiomusic-clicks/s-tfgHjwRkvpf

gogoha, 2020-11-03:
Same problem, i have tried with two quality SD cards

klendroid, 2020-11-27:
Having this exact issue as well. Here's a recording:
https://soundcloud.com/user-292410369/radiomusicnoise

Did you guys figure it out?

jacoblysgaard, 2020-11-27:
We did not!😭

klendroid, 2020-11-28:
Bummer! I have to say, I get the distortion all the time though, not just when pressing reset. It's always there.

klendroid, 2020-11-28:
I tried a different SD-card now, and tried to load in only the default RAW-files from the 8GB pack thing found here:
https://github.com/TomWhitwell/RadioMusic/wiki/Audio-for-the-SD-Card

... and now it works just fine! No distortion or glitching. :)

jacoblysgaard, 2020-12-09:
I tried just this, but even with the stock .raw files, I still get the same pops. Weird.

Morbo78, 2021-01-08:
I have the same problem and I'm using Sandisk and Kingston 32GB cards. It seems to be a bit better with the trigger input rather than the button, but it's still there. Tried altering crossfade settings, reinstalling the firmware and settings file, reflowing solder joints...

I'm using a Befaco PowSkiff power supply, but it also does it when just connected to my FC Power, too.

Links to audio files and photos of the boards here: https://drive.google.com/drive/folders/1TVoPL_2BiNpGv0wWosxHMdvPfK_WrFkP?usp=sharing

gogoha, 2021-08-16:
After format the card (Kingston 16Gb HC I U1 Class10) with "SD card formatter" app (overwrite format) and put the default RAW-files (uncompress directly on the sd card!!) that's work (2 hours testing).

MrMojoRisinGottaKeepOnRisin, 2023-11-06:
Getting a loud Pop/Click sound every time the **RESET** button is pressed and every time a signal is sent to the **RESET CV** jack. Using firmware version _"Radio_Music_2017.2.hex"_. Tried reformatting high quality microSD cards and also generic no brand name microSD cards. Tried several variations of the _"settings.txt"_ file but had the same results on three different RM modules. Reverted back to the old firmware _"RadioMusicA2.hex"_ and the Pop/Click sound is not present. Using older firmware is not a good solution for me because I would like to use all of the new features available with the latest version _"Radio_Music_2017.2.hex"_

An mp3 file containing four samples with Pop/Click examples is here -->
[**_http://llamamusic.com/docs/Radio_Music_Pop_and_Click_Problems.mp3_**](http://llamamusic.com/docs/Radio_Music_Pop_and_Click_Problems.mp3)