TomWhitwell/RadioMusic #154: Station/Reset CV sometimes selects wrong file

starmandeluxe, 2019-02-22:
I have two Radio Music modules (Rev 2.1 boards, latest 2017_2 firmware, loaded with WAV files) and both have this same issue: inputting a stepped CV (e.g. Turing machine or sequencer) into the Station CV jack does indeed change the channel, but sometimes it gets "confused" or "stuck" and fails to switch to the next channel and just plays the previous channel again, or just decides to play a completely random channel instead. Immediate is off for Channel CV so it waits for a trigger signal, which I clock to send at the same time as the Channel CV sequence. 

Basically, trying to sequence some sample changes consistently is impossible. I tried reducing the number of files in one bank from 16 down to 8 and finally down to 4, but the issue still happens even with just 4 files.

My Teensy version is 3.2 and my trigger CV has been tested with: Erica Synths Pico Trigger, Sputnik 5-step voltage Source Pulse OUT (variable pulse width, tried various widths).  I recorded a demonstration of these working correctly with a Pico Drum but the same CVs not consistently working correctly on both Radio Music modules:

[https://www.youtube.com/watch?v=yohYsuVO__o](https://www.youtube.com/watch?v=yohYsuVO__o)

Thonk-Steve, 2019-02-22:
Hi Alex,

A few questions:
What version of the firmware do you have loaded?
Did you build them yourself?
Where did you buy them?

Thanks

Steve

starmandeluxe, 2019-02-23:
Hello! These are from Thonk, firmware is the 2017_2 latest one I pulled from polyfather a couple of days ago.

Actually it looks like I was able to solve the problem via simply the settings file via this forum post suggestion:

`chanPotImmediate=1 
chanCVImmediate=1 
startPotImmediate=0 
startCVImmediate=0 `

I have no idea why this fixes the issue but all is well now. I'd be curious to know the logic behind it though.

starmandeluxe, 2019-02-23:
OK I did some more testing and think I understand why it sort of solves the problem. These settings do not rely on the Reset trigger CV at all. It changes the station when the CV changes. I am not sure why that would be more accurate than waiting for a Reset trigger. It seems the combination of the reset and the station CV isn't working completely. My guess still leans on the reset circuit path having some issue. I'll try doing some basic troubleshooting with those related parts.

Basically it is usable for triggering at the same rate as the CV change (constant change). One hacky workaround would be to make a "silent" track at the start to simulate "don't play on this beat".

starmandeluxe, 2019-03-02:
I essentially fixed this issue by modding the firmware. I added an updated hex file into my branch here ("RadioMusicResetCVFix.hex"):

https://github.com/starmandeluxe/RadioMusic/tree/patch-reset-cv/Collat eral/hex%20files/Beta 

Basically I just wait for a FALLING signal to trigger the reset so that it waits long enough to catch the station CV change.

Details here: https://www.muffwiggler.com/forum/viewtopic.php?p=3018268#3018268

It's not a perfect fix, but I'll close this issue since the module is now usable enough for me. Thanks!

closed 2019-03-02 by starmandeluxe