TomWhitwell/RadioMusic #156: High pitched noise on output

jcrbcn, 2019-03-08:
I can hear the sound file playing, but there's always a high pitch noise over the actual sound.

Here are two examples of the noises over the bird sounds from the "Blank" pack:

1. https://www.dropbox.com/s/c7f2eibqlica260/Radio%20Music%20Audio%20Out.aif
2. https://www.dropbox.com/s/kom5buqeoz7pxjh/Music%20Thing%20Audio%20Out%202.aif

LEDs, CV, reset button and pots work as expected. The audio doesn't. Tried it plugging CV, output through HEX MIX VCA... Also checked the components and current.

I assume that there are some issues related to mine, but I still haven't found a clear answer to this. 

Thanks

starmandeluxe, 2019-03-11:
Good timing that you posted this issue, since I've been having this problem for a while (since I built my Radio Music), but just started looking into it and [posted this thread](https://www.muffwiggler.com/forum/viewtopic.php?p=3023551) about it. Some others have reported this problem on the main build thread as well.

My issue is not as extreme as yours (which firmware are you using?) but the underlying problem may be the same. Also, how do you prepare your samples? Try normalizing to maximize the amplitude of your audio files. Also play around with the gain trimmer on the back of the module.

I haven't found a solution for this one yet (I don't think anyone has), but there are some steps you can take to get it to at least a usable state.

jcrbcn, 2019-03-27:
I've been using the 2017 v2 firmware. The samples I used were the ones from the "Empty SD Card" pack, so I assumed no normalization or bit depth change had to be done.

And yes, in pitch mode the knob changes the buzzing noise frequency.

elterotero, 2019-03-27:
I have seen the same issue also recently, did some reflowing, making sure the hardware was as it should be and after almost losing hope reformatted the Sd and copied the files back. It worked for a few days without the noise but now I can see that the noise is coming back again. 

Latest firmware and V2 board.

starmandeluxe, 2019-03-29:
> And yes, in pitch mode the knob changes the buzzing noise frequency.

That's interesting. The reason I asked is because it does the same for mine as well (try with a silent audio file). If that is the case, wouldn't it stand to reason that the buzz is introduced in the software, not the hardware? Although I don't know the full signal path, it is suspicious. Sorry I can't give more help, right now I have my module on low gain on the trimmer to minimize it, but I'm not sure it will help you with such an extreme case. Maybe also try the steps that @elterotero suggests.

Thonk-Steve, 2019-03-29:
The implementation of pitch on the Radio Music was very much pushing the
hardware in a direction quite far away from the original concept Tom
settled on with Radio Music, which was literally to fill the module with
lots of long form 'Radio like' audio that is constantly playing, and the
concept of tuning between those 'channels'.

So the pitch mode is certainly a bonus feature and it does reveal that this
module doesn't have a super low noise floor.

PZHRHG, 2019-05-13:
Ciao .. I use ADIO MUSIC mostly as DRUM Machine . I always have as FIRST FILE with 1sec of SILENCE. (sort=1) . first File is 00_SILENCE1SEC
that help me to switch between the BANKS real time while playing without having first file to fire audio and also allow some Chocke tecniques that I liked a lot on my DinkyTaiko.  I don't use PITCH Shift.

I also have a very audible High Pitch digital noise . and using Silnece file a lot it is very audible.
It is also very audible when I use Sines

This is very bad because so far to mask the issue I'm foced to play mostly Noise Samples (Haswell like)
but for more clean programs I might need to migrate to another module (like Tiptop ONE) , I really hope not.

Is there any improvement in the pipeline ? is this issue possible to solve ?

thology, 2019-08-08:
Having this exact same issue. Identical to your problem of that high pitched noise. Almost like a ground loop. Have there been any fixes for this?

Eliad-beep, 2020-04-30:
Hi all, 
I am experiencing the same issue as well. The background buzz is not as loud as the examples posted here but still loud enough to grab attention and disturb music. I almost lost hope with this one and am considering selling my RM (although I am having lots of fun with it). are there any new discoveries regarding this?

niall-munnelly, 2020-06-28:
+1 here, too.
There is a low red noise rumble when the module is idle, and a droning done with audible ticks during sample playback. The higher-pitched noise lasts as long as the sample plays and cuts abruptly when playback ceases. I’m using a modified firmware, but the behavior is the same.

Thonk-Steve, 2020-06-28:
If you want to upload audio samples of the noise we'd be happy to listen
and compare to what we know the noise of a normal radio music sounds like.
The ideal sound would start with a 0db tone, then have some silence and
then end with another tone.

Like all Eurorack modules you should experiment with where it is in your
case (far away from PSU is a good idea) and also keep it away from very
electrically noisy modules (worse culprits are modules with screens,
expensive 'conputer' modules and l
Those with lots of LEDs).

Also if you are using a switching type PSU, the cheap laptop type ones can
be very noisy. It's best to work out other strategies first though.

adamjav, 2020-09-04:
Hello all, 
I recently went through a similar issue. In my case it was absolutely a soldering problem on the back PCB with a capacitor in the audio path close to the DAC output. The soldering looked fine on inspection but the component was loose. I found it by recklessly "feeling around" the board while powered and outputting audio. Putting my finger in a certain spot would make it work normally. I powered it off, added a bit more solder. All good since then. Just wanted to put this forward as I'm an experienced technician and thought all was good on my board/ couldn't find the issue/ started looking at various posts and speculations on the topic that suggested it was a Teensy issue etc. etc. Then thought nah...hang on plenty of people not experiencing this... gotta be my issue. 
Regards, 
Adam

niall-munnelly, 2021-01-17:
> If you want to upload audio samples of the noise we'd be happy to listen and compare to what we know the noise of a normal radio music sounds like. The ideal sound would start with a 0db tone, then [...]

Here's mine. It's not quite what you asked for, and I'm sorry for that, but I think it demonstrates the behavior well enough:

https://www.dropbox.com/h?preview=Radio+Music+Troubleshooting.mp3

It starts with manual triggers, changing the sample. Then there are manual triggers, changing the pitch, and here you can hear the noise changing pitch as well. Finally there's just a quick pass with a sequencer.

This admittedly is something of a "Hail Mary" pass here; I'm just hoping someone will recognize this before I send it off to a tech.