TomWhitwell/RadioMusic #161: GND_F = Analog ground instead of digital ground

Mynasru, 2019-07-28:
The front panel only has analog ground, also for the SD card and LEDs. What is the reason behind this? Shouldn't the digital parts be connected to digital ground (DGND or teensy GND) instead of AGND?

Mynasru, 2019-09-19:
@TomWhitwell Could you explain this, I'm curious what is the reasoning behind it.

Mynasru, 2019-10-07:
> @TomWhitwell Could you explain this, I'm curious what is the reasoning behind it.

I wanted to make my own PCB and was checking your schematics. Paul (from teensy) is [saying](https://forum.pjrc.com/threads/110-GND-vs-AGND) AGND should only be used for analog circuits .

TomWhitwell, 2019-10-11:
Hi, sorry for the delay in replying. This was done to simplify the design. If you want to redesign with the extra pins and routing for the analog ground, you may get different results. I found that the shared ground gave results that I was happy with.

closed 2019-10-11 by TomWhitwell