TomWhitwell/RadioMusic #163: Low Volume, Trimpot At Back Does Nothing

angiamusic, 2019-08-26:
Built a new MTM Radio a while back, worked perfectly for some time, even used it at live gigs, but I recently moved to a new, bigger case and ever since then the output volume has been really low and turning the blue trimpot at the back does nothing. All CV inputs, Reset button and LEDs seem to be working ok, its just the output volume. 

For reference, this started when I moved to a new case with a new power supply, I've since resoldered some of the less perfect joints and replaced the TL074CN but nothing made any difference. The volume doesn't change regardless of plugging/unplugging any other CV into the module. Any help would be appreciated.

TomWhitwell, 2019-08-26:
That sounds very odd - if the trimpot does nothing, perhaps it is at fault. That could possibly explain this behaviour.

angiamusic, 2019-08-26:
Perhaps. I thought the same thing and re-soldered the trimpot itself to no avail. Maybe I need to change the trimpot entirely?

Thonk-Steve, 2019-08-26:
It might be worth trying that, maybe check also with a multimeter that
there isn't a short to ground on the output, or measure resistance between
the output signal and ground.

angiamusic, 2019-08-26:
> It might be worth trying that, maybe check also with a multimeter that there isn't a short to ground on the output, or measure resistance between the output signal and ground.
> […](#)

Thanks for the reply, I just checked and there's no short to ground on the output.

angiamusic, 2019-08-26:
Alright, I didn't have any potentiometer with me of that trimpot design at the moment, but I again redid the solder joints in a desperate quick fix attempt, and it works fine now! Thank you, Steve and Tom, for your help.

closed 2019-08-26 by angiamusic