TomWhitwell/RadioMusic #164: Low volume

oliverhd, 2019-11-03:
I have very low volume coming from chord organ and dont know why. have tried turning pot on the back but does not seem to change. the person i bought it from said he made it so when using the micro usb port for installing firmware it woulndt need eurorack power at the same time. dont know if this affects it, or if i somehow have exeeded the power limiting but other modules work fine.

Thonk-Steve, 2019-11-03:
Hi Oliver,

Do you have DIY skills? If it arrived like this I would send it back to the
person who built it and get them to fix it, this is a soldering issue and
has nothing to do with the USB port or software flashing.

You should hear big changes in volume on the output when turning that
trimmer on the back.

Regards

Steve
Thonk

oliverhd, 2019-11-03:
hmm. I have a little experience but will contact the person i bought it from. is the a way to see if the solder is bad? like oxidising?

Thonk-Steve, 2019-11-04:
Maybe send some photos of both sides of both PCBs?
I meant more that it suggests an issues with either a short circuit, a bad
solder joint, a poor connection between the two PCBs and the Teensy or that
a part has been damaged during the build.

oliverhd, 2019-11-04:
wow i fixed it myself, upon further inspection it was clear there was a lose connection since volume came back when pressing the panels a little together in the top. Then i looked for bad soldering or non at all first one of the power connectors had a little hole in the solder blob so i just touched it to heat up an form properly around the hole. then wich was probably the reason one of the resistors had very little if any solder on one of them and i gave it a little and now it works :D  thank you so much for your help

Thonk-Steve, 2019-11-04:
brilliant, well done :)