TomWhitwell/RadioMusic #166: Teensy 4.0 compatibility

Mynasru, 2019-11-12:
I was wondering if the radio music and chord organ hardware and/or software are compatible with the new Teensy 4.0, and if someone already tested this. I'm also wondering if anyone has any feature request or ideas for using the additional audio processing power of the Teensy 4.0 in the Radio Music module.

TomWhitwell, 2019-11-12:
Hi, the hardware isn't exactly compatible - I haven't checked the pin compatibility between 3.2 and 4.0, but I noticed at least that the DAC output has moved. The software would - I imagine - be compatible, but wouldn't take advantage of the extra speed or the SD card driver improvements that I think were introduced in 3.5/3.6.

Mynasru, 2019-11-17:
> Hi, the hardware isn't exactly compatible - I haven't checked the pin compatibility between 3.2 and 4.0, but I noticed at least that the DAC output has moved. The software would - I imagine - be [...]

After a bit more research I found out that there is [no DAC on the Teensy 4.0](https://forum.pjrc.com/threads/57208-No-DAC-on-4-0-Alternatives?p=212602&viewfull=1#post212602), so there is need for an external DAC (like PT8211).
As the hardware needs to change, software does too (aka new product). So the answer is the teensy 4.0 is fully incompatible with the Radio Music hardware. It would still be interesting to hear from others what features they would like to see when more processing power is available.
Maybe someday a v2 will come ;) Thank you for your help and great projects!

closed 2019-11-17 by Mynasru