TomWhitwell/RadioMusic #167: Random resetting/glitching

pavlidiselias, 2019-12-06:
This build was originally a chord organ but I decided to switch to a radio music firmware.
I am using it as a simple sample player triggering a kick on regular intervals.

About 2-3 times per minute the radio music will glitch and either reset or freeze for a second and then resume normally. Whilst this happens a sound is heard from the output sort of like static.
It never fails to glitch like this.

I don't know if this is a hardware issue or a software issue but it wasn't always like this.

Thonk-Steve, 2019-12-06:
Hi,

It's sounds to me like you have uncovered an issue with either the SD card
media or perhaps the soldering of the SD Card socket, these issues probably
wouldn't be found with the Chord Organ firmware.

Chord Organ doesn't constantly stream data from the card and if loading
from the card only downloads a tiny amount of data.

I'd suggest trying another SD card first.

Steve

pavlidiselias, 2019-12-06:
Hi Steve,

I'll try both of these individually and report back.

pavlidiselias, 2019-12-06:
Seems it was the SD that was causing the issue, used a different one and so far no glitches are present.

closed 2019-12-06 by pavlidiselias