TomWhitwell/RadioMusic #168: Teensy 4.0?

Spin2Win, 2020-03-26:
Wondering if anyone has tried a RadioMusic build with a Teensy 4.0.

I've got the panel and PCBs, and just put in my first component order.  I'm reading about the 4.0 and with similar pin-outs to the 3.2, I'm hoping it will just drop in.  

As well, with the faster speed, will it process .mp3s better?

Anyone have any comments?

Thonk-Steve, 2020-03-26:
It's not compatible, Teensy 4.0 has no hardware DAC and the software
compatibility isn't trivial either.

Spin2Win, 2020-03-26:
Thanks!  Ordering the 3.2 rev now.

Aq0amancer, 2021-12-11:
Will we be seeing a Teensy 4.0 version soon? With the Teensy 4 audio adapter, the DAC is available.

jojo-monk, 2023-03-05:
What about the compatibility of the teensy 3.6 ?

TomWhitwell, 2023-03-05:
Hi, the 3.6 certainly wouldn't fit in an existing Radio Music - the DAC pin would make it impossible to fit.

jojo-monk, 2023-03-05:
Maybe with a little strap ?