TomWhitwell/RadioMusic #174: Radio Music will not read SD Card, just fast reset each time

ApocPro, 2020-11-13:
I recently built a Radio Music.  I am having the same issue as electrojuke from 2018.  The module test works as expected, but with the Radio Music firmware loaded, it appears to be unable to read the SD card.  Without the SD card it slowly cycles through the LEDs and resets looking for an SD, but as soon as I put the SD card in it cycles the LEDs super quickly and resets the module with a loud click.  I have tried SanDisk Ultra class 10, class 4, 32 GB, 16GB, 8 GB.  All exhibit the same behavior.  Anyone have any ideas?

TomWhitwell, 2020-11-13:
Hi, sorry to hear that
- Check on the way the SD card is formatted - https://github.com/TomWhitwell/RadioMusic/wiki/SD-Card%3A-Format-and-File-Structure
- Check you have the latest firmware from here: http://www.polyfather.com/radio_music/
- Check soldering on the SD card and on the Teensy 
- You could try the Chord Organ firmware to see if that works, less dependent on SD card - might narrow down the issue: http://polyfather.com/chord_organ/

ApocPro, 2020-11-14:
I purchased a brand new SD card, SanDisk Ultra 32GB and added the files.  The previous SD cards I had been using had been used for something else before and I had reformated them using a Mac.  The erase function and Formatting for MS-DOS(FAT) seems to have been part of the issue.  After using a fresh SD as formatted from manufacturer, the Teensy wrote the settings as it was supposed to and hadn't been doing before.

It still had problems reading the files, so I reloaded the firmware as per your link above.  (The firmware that I had downloaded from the Github that was labelled as latest appears to be formatted for the Teensy 3.1 and I am using the 3.2).  After downloading Radio_Music_2017.2.hex everything is functioning as expected.

Thank you for your help and quick response.

Bassliner303, 2021-11-04:
My Radio Music has a similar issue. It simply does not read the SD Card. I tried 3 cards, SanDisk Extreme 32 GB, SanDisk Ultra 32 & 16 GB. No luck with that. Firmware is Radio_Music_2017.2.hex. Keeps resetting in a loop.

Chord Organ however is working.

Thonk-Steve, 2021-11-05:
If Chord Organ firmware is working and Radio Music isn't then that suggest
a possible soldering or mechanical issue with the SD card holder or the
connections between the card holder and the Teensy on the back. Checking
continuity with a multimeter and reflowing any suspect joints between the
SD card holder, through the various board connectors and to the teensy is
the best bet next. Chord Organ only has optional SD card requirements, it
works perfectly well without it working.

If you're using the latest Radio or Chord firmware from
http://www.polyfather.com/ then one good test to do to see if the SD card
can be written to (but perhaps can't be read from) is to hold down the
button while the module boots with an SD card in. If the Teensy is able to
write to the card then it'll put a file called settings.txt onto it. You'll
be able to see this file when you mount the SD card on your computer.

Bassliner303, 2021-11-05:
Thanks for your quick reply. I finally got it working. 

I checked the Gerber files, followed the traces and checked continuity with a multimeter. It had nothing to do with the card reader. There has been a bad trace on the pcb which connects the middle pin of the 7 pin connector to one leg of the station potentionmeter. Fixed that with a wire and now its working as it should :-)

closed 2023-04-05 by ApocPro