TomWhitwell/RadioMusic #176: Module works briefly after loading firmware then sound rapidly degrades

aaronstevenwhite, 2021-03-13:
I've looked through the issues reporting buzzing and distorted sound, and what I'm experiencing seems to be slightly different. I've replicated the following on both a PGH power supply and an Arturia RackBrute 6U power supply, both with the module alone on the supply and with other modules. 

After initially loading new firmware onto the teensy (I've switched back and forth between Radio Music, using the suggested sound library, and the Chord Organ), the module works perfectly for one or two minutes—station, start, and reset all function correctly, LEDs functions correctly, and I get undistorted sound (maybe with some hum) with both firmwares. Then, the sound begins to rapidly degrade into the sort of buzzing reported in a variety of other issues until the buzzing is the only sound that can be heard. When I turn station or press reset, it momentarily affects the buzz.

Powering down for a few minutes and then powering back up does not fix this issue. The only way to "fix" it is to reload the firmware, at which point the sound is perfect again for one or two minutes before degrading in the same way.

TomWhitwell, 2021-03-14:
Sorry to hear that, it sounds very odd. 
Have you done the 100k resistor fix? #70 https://github.com/TomWhitwell/RadioMusic/issues/70
It’s possible that a quick power cycle would leave the capacitor charged (distorting) while taking the module out and changing firmware would give it time to dissipate.

aaronstevenwhite, 2021-03-20:
Thank you for the extremely quick response! The resistor 100K fix seems to fix the issue modulo some remaining low-level hum. My apologies for creating a separate issue. 

I was under the mistaken impression that that fix was for an older version, and the apparent connection to reloading the firmware (and the short timescale compared to the hours before degradation others reported) made me think this issue was distinct. 

I'm still somewhat unsure why it seemed to be linked with swapping out the firmware. I did do quite a bit of testing before posting this issue involving waiting the same amount of time with and without swapping firmware, unplugging the module and letting it sit, and even plugging the module into USB but not swapping out the firmware; and the only thing that made a difference was swapping the firmware. Very odd.

Thanks again for the quick response and apologies again for the superfluous issue. I know how annoying responding to issues with documented fixes can be.

closed 2021-03-20 by aaronstevenwhite