TomWhitwell/RadioMusic #178: Prok as Radio Music crashes at seemingly random moments

Candymongo, 2021-06-14:
Somehow the Radio Music crashes at seemingly random moments. 

Sample:
https://imgur.com/gallery/aZC0ZTw

Amount of samples in total on the SD card is less than 300. The amount of samples in this particular bank is less than 30, might even be limited to 20. I am using a 16GB Ultra SD card. Could this be caused due to the SD card being to slow? Or is it perhaps a compatibility issue, that the Radio Music Teensy has issues with the hardware? On MW I got the response that there shouldn’t be an issue, so...

Thonk-Steve, 2021-06-14:
My gut reaction is bad SD card because this is an unusual problem.

Is the crash always like in the video?

Candymongo, 2021-06-14:
> My gut reaction is bad SD card because this is an unusual problem. Is the crash always like in the video?
> […](#)

Hey Steve, thanks for responding, appreciate it. I haven’t managed to find anything faulty with the SD card, but... Meanwhile tried to rule some stuff out. As that didn’t change anything I eventually tried a 32GB Extreme SD card and all seems to run pretty smoothly now.

As stated, the Ultra SD card didn’t show any faults during testing: Could it be the speed, or should the Ultra SD card suffice for what I tried in the video?

Thonk-Steve, 2021-06-14:
Hey, I don't think it's a speed thing, assuming they are both SanDisk cards
then the speed of either is probably vastly higher than required.

Candymongo, 2021-06-14:
Yeah both are Sandisks. I’ll stick with a few new Ultra’s then. Thanks Steve!

jimmymike0303, 2024-03-12:
So my Radio Music is doing this. I tried using a 32 GB Ultra, but that didn't work. I have reflashed the firmware as well. Any thoughts?

Candymongo, 2024-03-12:
> So my Radio Music is doing this. I tried using a 32 GB Ultra, but that didn't work. I have reflashed the firmware as well. Any thoughts?

Hey Jimmy, I'd advise you go to Prok Discord through:
 https://discord.gg/FXNjw4C they're very helpfull!