TomWhitwell/RadioMusic #180: Background noise during playback

Eliad-beep, 2021-09-26:
Hi all, 
I bought an assembled RM a couple of years ago and have had this issue since the start. I've tried figuring this out a few times and always failed. Maybe someone here can help. 
My module makes a consistent background noise when playing back files. any file. It is faint and in moments of silence (within the sample) it can be very clear. It also has a certain tonality to it so it can be quite distracting. 
Here is an example demonstrated by playing the "standard" birds sample that comes with the blank SDcard setup: https://soundcloud.com/eliad-wagner/normal-volume-birds-example/s-2K0ST5W6kbG?in=eliad-wagner/sets/rm-issue-debug/s-zaHeIke5ay1

Here is an example of one of my own personal files where I amplified the output quite a bit in order to make the issue clear:
https://soundcloud.com/eliad-wagner/example-1-via-rm/s-6fvKfJPDcg4?in=eliad-wagner/sets/rm-issue-debug/s-zaHeIke5ay1

The original file on the SDcard is a 16bit/41khz WAV file and you can hear it here for reference:
https://soundcloud.com/eliad-wagner/example-1-original-16bit-41k/s-4r1Q1nup0HA?in=eliad-wagner/sets/rm-issue-debug/s-zaHeIke5ay1

one thing I noticed (and that is why I decided to finally write here) is that when switching to another bank which contains samples of higher sample rate (48k in this case) the tonality of the disturbance seem to double in pitch. see example here:
https://soundcloud.com/eliad-wagner/example-2-via-rm/s-JYSSmNb5dpK?in=eliad-wagner/sets/rm-issue-debug/s-zaHeIke5ay1

The original (for comparison) is here: https://soundcloud.com/eliad-wagner/example-2-original-16bit-48k/s-XnYRVJdIRom?in=eliad-wagner/sets/rm-issue-debug/s-zaHeIke5ay1
** Please note that the original file in this example is encoded as 16bit/48khz and is also a stereo WAV file (not sure if that matters)

I find the behavior interesting because it possibly hints to the fact that the source of the problem is somewhere in the Teensy chip ? or the mechanism of reading the files?

I made an entire playlist of these example so you could have an easy repetitive listening if you like: https://soundcloud.com/eliad-wagner/sets/rm-issue-debug/s-zaHeIke5ay1

My module has a red rev2 PCB and also has a 100K resistor fix applied to it on the back side of the back PCB (as discussed in issue[#70](https://github.com/TomWhitwell/RadioMusic/issues/70)

Thanks in advance for your help

jaykrizzle, 2022-05-14:
hey there, did you (or others) ever found a solution for this problem?
I have the same behaviour and the red rev2 PCB version and from reading your message it seems that the 100k resistor  doesn't solve the issue, right?

Eliad-beep, 2022-05-15:
Not solved and perhaps widely ignored by makers and users alike. I tried 2 versions of radio music modules with different sim cards, file types, formats... It's always there

Eliad-beep, 2022-05-15:
100k resistor didn't solve it either.
I honestly started thinking that it's my ears that just pick up on it and others don't mind it. 
Sold my module and moved on.

jaykrizzle, 2022-05-15:
Thanks for your fast reply!
Yes, i know that some (broadband) noise is normal as in most digital samplers. I have no problem with that. But i doubt that the humming is the normal behaviour of the module… is it? To get a more accurate picture of it, I would be curious to hear from others if they 
A) didnt have the humming on their builds
or 
B) they have it but they simply not care about it?

And since adding the resistor doesnt help in this regard, is there something that we could do about it - like changing code? Or changing the diy/ hardware side of the module?

Vykaar, 2023-02-03:
I have two Radio Music modules with 2.1 pcbs and both caused a low level buzz when connected to other modules. The way I fixed that was to make some leads that only connected the tips of the jacks so no ground loop was being made. I have many Doepfer LC6 and LC9 cases and they both did this in any case but are now quieter if I only connect with single core groundless cables.