TomWhitwell/RadioMusic #183: Several CV Issues mainly Reset In

PeterSpiegel, 2022-06-10:
I bought a used MTM Radio Music and it functioned a bit strange. The CV in for Reset did not respond to any type of control voltage. The CV start did not seem to respond either. The only CV in that seems to respond is the station. The start position Knob also does not seem to respond and I can't select the banks holding the reset button down.

I updated the firmware to the 2017.2 hex file on a teensy 3.2 board and loaded the standard settings.txt. Unfortunately I still get the same behaviour.

Could it be a build issue and where would I be best approaching it. The only odd solder points are for the pins on the teensy. The other solder work on the 2 boards looks professionally done.

Any advice as to where I should best probe with a multimeter be mich appreciated.

TomWhitwell, 2022-06-10:
Hi, sorry to hear that. If the Teensy pins look dubious I'd start by resoldering those.

PeterSpiegel, 2022-06-10:
Thanks Tom, I'll start from there on in and maybe have a look into some of the resistors with a multimeter. I've seen some issues here with R18 and R19 for the reset in

PeterSpiegel, 2022-06-11:
[image]
[image]
[image]
[image]
[image]

Here are some Fotos of the solder points on the teensy. As you can see I don't think they are done very well. I think this would be a good point to start from. Also the female connector has a break at one point and I'm not sure if that could cause an issue