TomWhitwell/RadioMusic #186: Reset CV not working, Start CV behaving weird. Also sometimes glitching.

finnglink, 2022-10-24:
Picking up on #183 here, I've taken over from @PeterSpiegel.

I've yet again inspected the board. One of the electrolytic caps was the wrong way around. Fixed that, and reflowed all the joints.
The CV ins are now mostly working. I've also refreshed the latest firmware with a default settings file.

- CV Station is fine
- CV Start works, but on short samples I can go beyond the sample it should play (using a +10V Befaco Joystick as source)
- Reset button, LED and bank change works.
- Reset CV in however, does nothing. I've already swapped the two BAT42s and went through it with a multimeter. I get continuity all the way through, from the jack to the pin on the Teensy. But still, nothing is happening when I plug a gate in there (I've tried 5V and 7V, but I assume 3.3V should be enough to work)

Also, sometimes the sounds seem to glitch out when I feed a CV in there. They still somewhat respond to pots and CV, but can only go back to normal on a power cycle.

Any ideas on where to look next? Could the TL074 be faulty (causing the glitch-hang-up and weird CV behavior), or are some of the Teensys inputs toast? 

Best,
Finn

finnglink, 2022-10-24:
This is what I'm at right now:
[image]
[image]
[image]

finnglink, 2022-10-24:
Update 2022.10.24 / 15:45:

Swapping the TL074 seems to have fixed the sample jumping on higher start-cv voltages.

With a little code-snippet:
` digitalWrite(11, digitalRead(9)); `
I could verify, that the Reset CV is fully functional, and that it seems to be a software problem.

I've also tried a newly formatted (FAT on Mac, I can only choose between ExFAT and FAT) 32GB SD card with just two samples doesn't change anything. I'm currently on the 2017 Firmware **(Radio_Music_2017.2.hex)**. The older version **RadioMusic1_1.cpp.hex** doesn't work at all. Without an SD card I get the four lights blinking at the same time, with the card inserted it's completely dead.

CallPhysical, 2023-01-29:
Did you ever get it working? I just bought one used and it has the same rev board as yours, but there are two extra resistors that have been added on the back of boards as a fix for something. I wonder if yours needs the same.

finnglink, 2023-01-31:
> Did you ever get it working? I just bought one used and it has the same rev board as yours, but there are two extra resistors that have been added on the back of boards as a fix for something. I wonder if yours needs the same.

I did not, I gave up haha. I don't really have the time for DIY at the moment, so I passed it on to someone from our Discord. Would you take a picture of the resistors I can share with them?

Thonk-Steve, 2023-01-31:
The resistors thing is most likely to resolve a distortion issue that some
people would have when the module is left on for a long time.

It sounds to me like you possibly have a short circuit that's feeding into
the CV system somewhere, did you get your kit from Thonk?

finnglink, 2023-01-31:
> It sounds to me like you possibly have a short circuit that's feeding into the CV system somewhere, did you get your kit from Thonk?
> […](#)

Thanks Steve!
I bought it as damaged / failed DIY, I don't know where it's from originally.