TomWhitwell/RadioMusic #193: firmware problem

pottermcmuck, 2025-05-09:
I have a used Radio Music that I wanted to reset. It was working when I got it. 

I tested the Teensy with the slow and fast blink, all was good. 

I installed the 2017.2 firmware and everything seemed fine. 

I loaded the blank SD with the bird and vinyl sounds, everything was good.

I tried to load the suggested audio bundle, and here's where things went sideways. It didn't load, I kept getting 100060 error messages, which usually means that either my computer or the SD card are the problem, but I reformatted the card and everything seemed ok. I was able to load the suggested audio bundle.

Unfortunately, since then the module has stopped working. 

I've retested the teensy and it's fine. But when I load the SD nothing works, even the blank SD samples (birds & vinyl). 

I tried installing the Chord Organ firmware and it worked fine. I've tried retesting and reinstalling several times, nothing changes. 

According to the manual, this kind of behaviour means it's a firmware problem. But I'm not sure. 

Please help!

pottermcmuck, 2025-05-10:
Update: After installing the firmware several times, as well as older firmwares, and reformatting the SD card a few more times and loading it with various files, the module somehow managed to work again using the latest firmware (2017.2), with the birds & vinyl blanks loaded on the SD card. However, I'm not able to put anything else onto the card, so I now suspect the card to be the problem. I'll buy another card as soon as possible and try this process again.

TomWhitwell, 2025-05-10:
Hi, sorry you're having difficulty - it sounds like the issue is in the file structure of the SD card, rather than anything else

cabb-ui, 2025-11-13:
The link to the 2017.2 firmware is broken ... can you upload it ? none of the other versions seems to work for me

closed 2025-11-14 by TomWhitwell